le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #1: TMC2209 UART?

Miraculix200, 2020-05-06:
So, how does the TMC2209 UART connection look like on the hardware side?

You connect RX/TX on the 2209 together, and then connect them to a dedicated pin for each stepper driver on the board, right? Did you change the hardware address of the stepper drivers? Did you use a resistance between RX/TX and the pin on the board? Did you cut the RX/TX pins on the bottom side of the stepper module (because according to the schematic of the Robin Nano 1.2 board, the RX pin is connected to the CLK pin via the board)

Miraculix200, 2020-05-06:
I think I just screwed up because I did cut the RX/TX pins on the bottom side. Getting TMC CONNECTION ERROR on the display. Will be fun to put this thing together again

le3tspeak, 2020-05-06:
Q: You connect RX/TX on the 2209 together, and then connect them to a dedicated pin for each stepper driver on the board, right? 

A: Yes PA3 ->X, PA6->Y, PA1->Z

Q:Did you change the hardware address of the stepper drivers? 
A: No Default

Q:Did you use a resistance between RX/TX and the pin on the board? 
A: No, One Single Wire

Q: Did you cut the RX/TX pins on the bottom side of the stepper module (because according to the schematic of the Robin Nano 1.2 board, the RX pin is connected to the CLK pin via the board).

A: No, Only Remove All Jumpers for the Driver

Q:Also, what needs to be done on the software side? Simply set the drivers to TMC2209 (instead of TMC2209_STANDALONE) and uncomment the lines in pins_MKS_ROBIN_NANO.h? Or is there somethign else to be done?

A: Yes and TMC Baudrate

Miraculix200, 2020-05-06:
Thanks. Will try again with the jumpers removed. But I still think I screwed it up haha

Miraculix200, 2020-05-06:
Actually it looks like it worked. Even with the pins on the bottom side removed. "M122 V" only says "Recv: Testing E connection... OK", but I guess that's normal. Or should it be more verbose?

le3tspeak, 2020-05-06:
Enable #define TMC_DEBUG for More Informations an Retry M122

Miraculix200, 2020-05-06:
Will do. Did you also try if it works with the PA9 and PA10 pins from the wifi module slot? Because some people don't want to use the pins on the 5th stepper slot.

le3tspeak, 2020-05-06:
PA2, PA9, PA10, PB2, PE6 will Work

closed 2020-05-06 by le3tspeak

EPDCenter, 2020-05-06:
Gracias por responder, qué puente debo eliminar del controlador?https://es.aliexpress.com/item/33028054001.html?spm=a2g0s.9042311.0.0.347a63c0JnZiGQ

EPDCenter, 2020-05-06:
How would the jumpers go if they did it using slave_addrees?

  /**
   * Four TMC2209 drivers can use the same HW/SW serial port with hardware configured addresses.
   * Set the address using jumpers on pins MS1 and MS2.
   * Address | MS1  | MS2
   *       0 | LOW  | LOW
   *       1 | HIGH | LOW
   *       2 | LOW  | HIGH
   *       3 | HIGH | HIGH

le3tspeak, 2020-05-06:
FYSTEC V3 is not designed for 1 cable connection 2 cables per driver FYSTEC V2.1 is therefore not feasible for 1 cable connection for all stepper drivers because there are not enough pins available

EPDCenter, 2020-05-06:
> FYSETC V3 no está diseñado para 1 conexión de cable 2 cables por controlador FYSTEC V2.1, por lo tanto, no es factible para 1 conexión de cable para todos los controladores paso a paso porque no hay suficientes pines disponibles

How could I do it using Slave address? How would the jumpers be placed? : S

le3tspeak, 2020-05-06:
Meintest du: Ich weiß nicht was du mit dem Serail machen möchtest ab das Robin geht nicht mit HW Serial für UART nur Software Serial und da sind die Jumper nicht nötig da alles Per Firmware eingestellt wird
193/5000
I do not know what you want to do with the serial from the Robin does not work with HW Serial for UART only software serial and there are no jumpers because everything is set by firmware

EPDCenter, 2020-05-06:
Check it please : https://www.youtube.com/watch?reload=9&v=ZK9ttGcpiC4

le3tspeak, 2020-05-06:
YES as I said before he also uses software serial and no HW serial everything is set in the FW and not by jumper and a 2 cable connection is required without resistance and he uses PA6 for TX and PA1 for RX an X to 0 Y on 1 on 2 and E on 3

EPDCenter, 2020-05-07:
How are such ports selected?