le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #10: power loss recovery doesn't work the recovery file is written to the sd but after the power comes back there is no summary menu

ivograca, 2020-05-26:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

le3tspeak, 2020-05-26:
No Configuration.h and Configuration_adv.h files added.
Which version?

ivograca, 2020-05-27:
my mistake. already updated with all the information thanks for your work

ivograca, 2020-05-27:
> No Configuration.h and Configuration_adv.h files added.
> Which version?

my mistake. already updated with all the information thanks for your work

le3tspeak, 2020-05-27:
MKS Robin Nano is not compatible with the function in the currently known bug in 2.0.5.3

closed 2020-05-27 by le3tspeak