le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #7: [BUG]  Repetier print don't work

horego, 2020-05-26:
Hello,

first of all i like your project. I decided to move from stock twotrees firmware to your marlin.
But now i can't print through Repetier. I don't know why. But it simply got stuck after the heatbet successfully reached it's target temperature.

With the stock twotrees firmware i had no problems. I tried to change the baudrate but it doesnt work (M575 command always show "echo:?(B)aud rate implausible."). I tried:

Baud rate commands:
> 19:30:48.711: N13 M575 S115200
19:32:50.445: N17 M575 S57600

Log of repetier:
> 19:37:06.165: N18 M73 P0 R17 Q0 S17
19:37:06.168: echo:Unknown command: "M73 P0 R17 Q0 S17"
19:37:06.179: N19 M140 S60
19:37:06.179: N20 M105
19:37:06.179: N21 M190 S60
19:37:06.179: N22 M104 S210
19:37:06.179: N23 M105
19:37:06.179: N24 M109 S210
19:37:06.182: N25 M82 ;absolute extrusion mode
19:37:06.182: N26 G28
19:37:06.189: Error:Line Number is not Last Line Number+1, Last Line: 20

Picture: https://ibb.co/TrzZChB

Firmware Version: Sapphire PRO - Stock 2.0.5.3 - EEPROM Release
Repetier Version: Repetier-Server Free 0.93.1

any suggestions?

Best regards

le3tspeak, 2020-05-26:
Try M575 P3 S115200 or If the printer is still completely unchanged, try the version here
[Robin_nano35.zip](https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/files/4684179/Robin_nano35.zip) Newer Release of the EEPROM Version

horego, 2020-05-26:
@le3tspeak Thx for your fast reply i already cloned the repo and compiled with 

#define BAUDRATE 115200

now it seems to be working...

le3tspeak, 2020-05-26:
The release files are 20 commits after I have to update them again ...

le3tspeak, 2020-05-26:
@horego please let me know if everything is ok with the newly cloned git now with the connection

horego, 2020-05-26:
@le3tspeak i printed my bed leveling test several times. With the value of 115200 everything works like before. This was the default value of the stock fw from twotrees. Maybe you can change it to in your repo as well?

By the way. Also tried the firmware with MKS Style V1 from the release section. After successfuly flashing the firmware there was only a black screen. I tried flashing the firmware to times. But this seems to be another issue.

le3tspeak, 2020-05-26:
@horego The MKS Style V1 needs the Robin-Nano35.bin and the folders on the UI Folder in the Root oft the SD

horego, 2020-05-26:
I know everything including the res. files flashed successfuly.

closed 2020-05-26 by le3tspeak