le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #9: bilinear mesh doesn't work don't compensate

ivograca, 2020-05-26:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

le3tspeak, 2020-05-26:
No Configuration.h and Configuration_adv.h files added.
Which version?

ivograca, 2020-05-27:
i have edit and update informations thanks for your work

le3tspeak, 2020-05-27:
Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
Connect to your printer from host software such as Cura, Printrun or Repetier Host.
Issue the command M111 S247 to enable maximum logging.
Perform a G28 to do your standard homing procedure.
Do a G29 to probe the bed. This will also enable bed leveling.
Copy the log output into a .TXT file and attach it to your next reply.
Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

le3tspeak, 2020-05-27:
Please be so good and try the Branch MKS Robin Nano I have revised the whole code and from now on only for the branch support otherwise it will be too confusing for me and your error should not happen anymore

https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/tree/MKS-Robin-Nano

closed 2020-05-27 by le3tspeak