le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #11: [BUG] [QUESTION] Print abort with large blob before end of print

horego, 2020-05-27:
Hello again,

first of all i'm not sure how this can happen. But i reproduced the problem several times. Some of them with other cura options and removed end of print gcode. But it all behaves the same.
 
While printing my bed level test the print aborts before it finishes with a large blob of filament.
When prinding with repetier everything is fine. When printing via SD Card this problem occures.

Used latest version of you eeprom branch.
GCode of bed leveling test: https://pastebin.com/QLmaTaVU
Picture of the unfinished print: https://ibb.co/L6cxmS2
STL File: https://www.file-upload.net/download-14105386/Bed_calibration_test.stl.html
Slicer: Cura 4.5.0

Any idea?
Best regards

le3tspeak, 2020-05-27:
@horego do you have in the endcode of the slicer M18 or M84?

horego, 2020-05-27:
@le3tspeak i removed M18 and M84 from gcode but the problem still exists. Why should i change a buffer size? It's only a buffer :) Maybe the buffer is being flushed before the print finishes?
The blob is now on another coordinate :D

proof: https://ibb.co/nR9k1pD

le3tspeak, 2020-05-27:
with M400 before the M codes at the end the same result?

horego, 2020-05-27:
Puh. I thought everything is getting better with marlin. Have you made any changes to the original marlin fw of the sd card stuff?

I cut the raw code to this one at the end of gcode:
> G1 F2700 E218.78572
M400 
M140 S0
M104 S0

I will edit this post in a few minutes

le3tspeak, 2020-05-27:
If M400 doesn't help, I still have an approach before I can test myself. I'm still at work.
Change:
`//
// SD Card
//
#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION              ONBOARD
#endif

#define SDIO_SUPPORT
#define SD_DETECT_PIN                       PD12
#define ONBOARD_SD_CS_PIN                   PC11`

too

`//
// SD Card
//
//#ifndef SDCARD_CONNECTION
//  #define SDCARD_CONNECTION              ONBOARD
//#endif

#define SDIO_SUPPORT
#define SD_DETECT_PIN                       PD12
//#define ONBOARD_SD_CS_PIN                   PC11`

horego, 2020-05-27:
I tried to increase the retry number. Without success. Maybe you can have a quick test when you come at home today.

le3tspeak, 2020-05-27:
Please be so good and try the Branch MKS Robin Nano I have revised the whole code and from now on only for the branch support otherwise it will be too confusing for me and your error should not happen anymore

https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/tree/MKS-Robin-Nano

le3tspeak, 2020-05-27:
Please be so good and try the Branch MKS Robin Nano I have revised the whole code and from now on only for the branch support otherwise it will be too confusing for me and your error should not happen anymore

closed 2020-05-27 by le3tspeak

horego, 2020-05-27:
Thx. I'll give it a try tomorrow.
From which github project / branch / commit is your codebase from?