le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #14: [QUESTION] Codebase, Branching and Commithistory

horego, 2020-05-28:
Hey,

can u please explain on which code your different branches are based on? I'm getting more and more confused. Five different branches... Hard reset over time... "Initial commit". What can i expect from the branches? Is there a stable and develop concept behind?

Thx

le3tspeak, 2020-05-28:
Hi,

Branch|Base|Distribution Date|Support
---|---|---|--
MKS-Robin_Nano | bugfix-2.0.x | 2020-05-28 | YES
2.0.X-UI | Version 2.0.5.3 | 2020-03-31 | NO
2.0.5.3-EEPROM | Version 2.0.5.3 | 2020-03-31 | NO
2.0.X-bugfix | bugfix-2.0.x | 2020-04-15 | NO

in the future it will only be on the branch: MKS-Robin-Nano go on with all other versions I will not go on but since there are people who use the versions that stay online in README.md you can see the differences between the versions and the hard reset on the MKS was because I totally messed up the code in Conf.h

horego, 2020-05-29:
Thx. and you forked the code of your branches from another mks branch. Which branch / date / commit is your codebase from? Which one from https://github.com/makerbase-mks when is your source?

When i flash the mks robin nano branch. It shows up to be marlin 1.x not 2.x. What is the truth?

le3tspeak, 2020-05-29:
The only version in the git that is related to MKS is the UI version everything else are original from the Marlin Github. The version number at the boot is my version number, which makes it easier for me to support, because it allows me to better determine the error whether the version used has not yet been revised or the error still exists.

horego, 2020-05-29:
Thx. For simplicity. Can u give a direct link to the MKS/Merlin repo branch/commit? It is hard to determine the source. Because you have not forked yor branches.

le3tspeak, 2020-05-29:
MKS Base only on the UI version, all others direct from Marlin. Distribution date is at the top ...

https://github.com/makerbase-mks/MKS-Robin-Nano

horego, 2020-05-29:
Ok. So it is https://github.com/MarlinFirmware/Marlin/commit/0ea26d538bfb883bed57c005a6c0c3a365c31acc
Are you planning to merge to a newer distribution? This is the last question. I simply like to understand what features/bugs and fixes are included in your code compared to the origin.

le3tspeak, 2020-05-29:
as already written I will only continue with the version MKS-Robin-Nano Version and the current state of the art of Marlin, all adjustments from me can be found in the other versions except the Eeprom Spi save which causes problems in the version therefore it was not taken over. A brief overview is that the TMC UART is fully usable, like Neopixel, such as fast switching between Pro and Plus, arc welding adjustments and much more ...

closed 2020-05-29 by horego