le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #19: [BUG] Judder resulting from usage of JD, please revert to Classical Jerk

thierryzoller, 2020-06-09:
Printer: Sapphire PRO
**Proposed solution:** Please revert back to classic Jerk, the only jerk control that works as it stands.
Link to Marlin Bugs/Issues :

Printing circles results in quite severe XY Judder ruining the print.
Print the attached shows in the first layers (big circle)
[judder_test_initial layers.zip](https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/files/4753563/judder_test_initial.layers.zip)

thierryzoller, 2020-06-09:
https://github.com/MarlinFirmware/Marlin/issues/18186

qwewer0, 2020-06-09:
Did you try the latest bugfix version?

le3tspeak, 2020-06-09:
The G code is not an everyday situation, however, this has been extremely exaggerated. In Cura, set the values for maximum resolution, movement resolution, deviation and the problem is gone in your JD setting, and here is a better example for printing circles, the original one Gcode where you linked has 6894KB and a correctly issued GCode for the firmware has only 741KB and the stuttering no longer exists try the G code yourself
[judder_test_initial_layers.aw.zip](https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/files/4754319/judder_test_initial_layers.aw.zip)

closed 2020-06-09 by le3tspeak

thierryzoller, 2020-06-09:
I am not sure I understand what your point is. You say this is not an "everyday situatuion". That's exactly what it is. The attached is a mould of mine and these are circular shaped. So this _is_ an everyday problem.

I am also not sure what you mean by "a correctly" sliced. I used Cura to slice with standard settings. Maybe you can explain what you mean by "correctly" sliced?

If that's the solution you should tell the Marlin leads about it, they think this can be fixed and thinkyhead was saying that we shouldn't have to lower the slicer resolution for it to print correctly.

I am seeing no solution and yet you just closed a valid report. Can you either explain what the solution is (what resolution is working?) or build without JD ?

thierryzoller, 2020-06-09:
Also please explain what the hell "correctly issued GCode for the firmware" is supposed to mean.

qwewer0, 2020-06-09:
If you slice the same modell with PrusaSlicer, then the jitter most likely won't be present. It is big part a slicer issue.

le3tspeak, 2020-06-09:
In the G codes you added there are circular movements of less than 0.075mm segment length, it is no wonder that nothing will work, especially with the Nano at 72Mhz Core Speed.
thinkyhead said the resolution settings don't match the JD, it doesn't work the way it should.
an example of a resolution setting used by an Ultimaker 3:

Maximum resolution: 0.8333
Maximum motion resolution: 0.75
Maximum deviation 0.05

which I also took on at the FW.

The firmware is optimized for G2 / 3 movements and you try to implement a circle with just one circular movement with the GCode here:
; TYPE: WALL-OUTER

G1 F2400 X120.417 Y126.562 E935.2212
G1 X119.908 Y126.897 E935.23742
G1 X119.393 Y127.218 E935.25356
G1 X118.863 Y127.528 E935.2699
G1 X118.335 Y127.818 E935.28593
G1 X117.79 Y128.099 E935.30225
G1 X117.243 Y128.364 E935.31842
G1 X116.689 Y128.614 E935.33459
G1 X116.134 Y128.847 E935.35061
G1 X115.567 Y129.067 E935.36679
G1 X114.995 Y129.273 E935.38297
G1 X114.418 Y129.463 E935.39913
G1 X113.83 Y129.639 E935.41546
G1 X113.248 Y129.797 E935.43151
G1 X112.658 Y129.942 E935.44767
G1 X112.064 Y130.069 E935.46384
G1 X111.461 Y130.183 E935.48017
G1 X110.867 Y130.28 E935.49618
G1 X110.263 Y130.36 E935.51239
G1 X109.654 Y130.426 E935.52869
G1 X109.054 Y130.474 E935.54471
G1 X108.442 Y130.507 E935.56101
G1 X107.834 Y130.524 E935.5772
G1 X107.231 Y130.525 E935.59324
G1 X106.623 Y130.51 E935.60943
G1 X106.011 Y130.478 E935.62573
G1 X105.411 Y130.432 E935.64174
G1 X104.801 Y130.367 E935.65807
G1 X104.202 Y130.29 E935.67414
G1 X103.601 Y130.194 E935.69033
G1 X103 Y130.082 E935.7066
G1 X102.405 Y129.956 E935.72278
G1 X101.82 Y129.815 E935.73879
G1 X101.233 Y129.657 E935.75497
G1 X100.644 Y129.482 E935.77132
G1 X100.066 Y129.294 E935.78749
G1 X99.499 Y129.092 E935.8035
G1 X98.927 Y128.871 E935.81982
G1 X98.371 Y128.639 E935.83585
G1 X97.811 Y128.388 E935.85218
G1 X97.267 Y128.128 E935.86822
G1 X96.727 Y127.85 E935.88438
G1 X96.189 Y127.557 E935.90068
G1 X95.663 Y127.251 E935.91688
G1 X95.146 Y126.932 E935.93304
G1 X94.641 Y126.602 E935.94909
G1 X94.138 Y126.254 E935.96537
G1 X93.647 Y125.893 E935.98158
G1 X93.171 Y125.526 E935.99758
G1 X92.696 Y125.137 E936.01391
G1 X92.236 Y124.74 E936.03008
G1 X91.785 Y124.331 E936.04628
G1 X91.351 Y123.914 E936.06229
G1 X90.919 Y123.478 E936.07863
G1 X90.503 Y123.035 E936.0948
G1 X90.1 Y122.58 E936.11097
G1 X89.707 Y122.116 E936.12715
[... 175 lines cut ...]
G1 X123.262 Y124.286 E938.9737
G1 X122.809 Y124.7 E938.99003
G1 X122.354 Y125.095 E939.00606
G1 X121.884 Y125.481 E939.02225
G1 X121.709 Y125.617 E939.02814
G0 F2160 X121.404 Y125.854
G0 X120.91 Y126.219

and in the file added by me, which is adapted for the FW, the many G code looks like a circular movement from you:
; TYPE: WALL-OUTER
G2 X101.444 Y114.965 I-8.725 J4.074 E43.79026 F2400
G2 X116.293 Y103.870 I5.868 J-7.630 E45.08598
G0 F2160 X116.039 Y103.255
M204 S5000
M205 X30 Y30
G0 F6000 X115.947 Y103.078
G0 X116.415 Y102.835
G1 F3000 E39.08598
G0 F6000 X118.073 Y108.003
G0 X116.021 Y107.872
G0 X129.041 Y117.833
G0 X127.159 Y116.961
G0 X126.821 Y118.077

you might notice the differences too, right?

le3tspeak, 2020-06-09:
and a second example of customized G code for the FW here is the first skirt line of your G code:

;TYPE:SKIRT
G1 F3000 E0
G1 F2400 X109.492 Y81.115 E0.03041
G1 X110.29 Y81.188 E0.07199
G1 X110.87 Y81.255 E0.10229
G1 X111.676 Y81.372 E0.14454
G1 X112.358 Y81.489 E0.18045
G1 X112.933 Y81.604 E0.21087
G1 X113.753 Y81.79 E0.2545
G1 X114.379 Y81.95 E0.28802
G1 X114.978 Y82.119 E0.32032
G1 X115.696 Y82.342 E0.35933
G1 X116.434 Y82.594 E0.39979
G1 X116.951 Y82.785 E0.42839
G1 X117.725 Y83.095 E0.47165
G1 X118.264 Y83.329 E0.50214
G1 X118.951 Y83.647 E0.54141
G1 X119.617 Y83.977 E0.57998
G1 X120.122 Y84.244 E0.60962
G1 X120.818 Y84.636 E0.65106
G1 X121.377 Y84.971 E0.68488
G1 X121.996 Y85.364 E0.72292
G1 X122.546 Y85.734 E0.75732
G1 X123.213 Y86.209 E0.7998
G1 X123.642 Y86.533 E0.8277
G1 X124.28 Y87.041 E0.87001
G1 X124.776 Y87.457 E0.9036
G1 X125.285 Y87.91 E0.93895
G1 X125.823 Y88.411 E0.9771
G1 X126.275 Y88.856 E1.01001
G1 X126.792 Y89.392 E1.04865
G1 X127.292 Y89.937 E1.08702
G1 X127.685 Y90.392 E1.11822
G1 X128.178 Y90.991 E1.15847
G1 X128.584 Y91.513 E1.19278
G1 X128.986 Y92.059 E1.22796
G1 X129.44 Y92.71 E1.26914
G1 X129.761 Y93.195 E1.29932
G1 X130.154 Y93.83 E1.33807
G1 X130.523 Y94.462 E1.37604
G1 X130.862 Y95.075 E1.41239
G1 X131.126 Y95.588 E1.44232
G1 X131.503 Y96.367 E1.48723
G1 X131.751 Y96.916 E1.51848
G1 X132.019 Y97.554 E1.55439
G1 X132.283 Y98.229 E1.59199
G1 X132.508 Y98.856 E1.62656
G1 X132.755 Y99.607 E1.66758
G1 X132.931 Y100.193 E1.69932
G1 X133.128 Y100.916 E1.7382
G1 X133.274 Y101.517 E1.77029
G1 X133.447 Y102.314 E1.81261
G1 X133.562 Y102.927 E1.84497
G1 X133.677 Y103.638 E1.88234
G1 X133.762 Y104.272 E1.91553
G1 X133.848 Y105.064 E1.95687
G1 X133.895 Y105.644 E1.98706
G1 X133.939 Y106.454 E2.02915
G1 X133.957 Y107.106 E2.06299
G1 X133.958 Y107.807 E2.09936
G1 X133.942 Y108.461 E2.13331
G1 X133.901 Y109.269 E2.17528
G1 X133.856 Y109.848 E2.20542
G1 X133.773 Y110.64 E2.24674
G1 X133.688 Y111.28 E2.28023
G1 X133.575 Y111.994 E2.31774
G1 X133.464 Y112.597 E2.34955
G1 X133.293 Y113.401 E2.3922
[... 163 lines cut ...]
G1 X103.244 Y81.385 E8.33106
G1 X104.039 Y81.267 E8.37276
G1 X104.63 Y81.196 E8.40364
G1 X105.422 Y81.122 E8.44492
G1 X106.012 Y81.082 E8.4756
G1 X106.767 Y81.051 E8.51481
G1 X107.457 Y81.04 E8.55061
G1 X108.146 Y81.048 E8.58636
G1 X108.241 Y81.052 E8.5913
G0 F2160 X108.907 Y81.078
M204 S5000
M205 X30 Y30

and here adapted to the FW:
; TYPE: SKIRT
G1 F3000 E0
G3 X132.282 Y98.228 I-1.400 J26.415 E1.59198 F2400
G3 X125.890 Y126.525 I-24.788 J9.269 E3.18496
G3 X97.275 Y131.905 I-18.389 J-19.021 E4.78457
G3 X81.043 Y107.894 I10.232 J-24.409 E6.37586
G3 X96.671 Y83.358 I26.454 J-0.395 E7.97402
G3 X108.240 Y81.052 I10.810 J24.063 E8.59130
G0 F2160 X108.907 Y81.078
M204 S5000
M205 X30 Y30

thierryzoller, 2020-06-10:
Thank you, that explains a lot. I do not know of any slicer that produces G2/G3 moves. Which one should we use (or supports arc) ?

As to Maximum Resolution : If I set Max Resolution to your settings above, the stutter will be less?

thierryzoller, 2020-06-10:
> If you slice the same modell with PrusaSlicer, then the jitter most likely won't be present. It is big part a slicer issue.

I also noticed this but didn't know why this is the case.

thierryzoller, 2020-06-10:
I resliced it with the CURA profile you uploaded, its 6MB and the settings are

-Maximum Travel Resolution iss 0.88
-Deviation set 0.05

Is that going to reduce the issue?

thierryzoller, 2020-06-10:
The settings you listed `Maximum resolution: 0.8333
Maximum motion resolution: 0.75
Maximum deviation 0.05`

Are not in the CURA profile you uploaded, is this an error?

thierryzoller, 2020-06-12:
For those having the same problem. I disabled JD enabled, Classic Jerk and S-Curve and the problem is gone. One notices a lot of small movements but no judder.

thierryzoller, 2020-06-17:
update: https://github.com/MarlinFirmware/Marlin/issues/17342#issuecomment-645493437

When using the newest bugfix-2.0.x, you can now change the config parameter

#define JD_HANDLE_SMALL_SEGMENTS
in Configuration.h to

//#define JD_HANDLE_SMALL_SEGMENTS
and thereby deactivate the code which is causing most (all?) of the stuttering and surface blobs. This is a workaround until an actual fix may be implemented, which - I guess - is at least a couple of months away.

le3tspeak, 2020-06-18:
https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/commit/41f88f707eebcb65566a8c9d07b478c435780168