le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #20: [BUG] Enabling power loss recovery freeze the printer!

enthropy92, 2020-06-11:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

le3tspeak, 2020-06-11:
It does not work with the Nano it is also not without reason (Creality3D) in the top of the Power Loss

closed 2020-06-11 by le3tspeak

enthropy92, 2020-06-11:
Ok thank you :)