le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #21: [BUG] No boot with SD in and Ru lang.

new1ru, 2020-06-14:
### Bug Description
Machine won't go past welcome screen if fitted with an SD card. 

### My Configurations
1.0.5a, lang set to RU, tried both Japanese an Cyrillic charsets with no luck
[config.zip](https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/files/4776423/config.zip)

### Steps to Reproduce
1. Flash the machine
2. Wait for it to load
3. Fail miserably (you're stuck)

**Expected behavior:**
Status screen shown after load.

**Actual behavior:**
Welcome screen with credits, no input on-screen buttons reaction, no status screen.

#### Additional Information
It does flash and then load after reboot if no card fitted, Russian text works fine. SD card can be then inserted and initialized manually in the menu, seems to work. Rebooting the printer starts everything again. The card was formatted clean, doesn't help.
Compiles with no errors.

le3tspeak, 2020-06-14:
Hi I'll have a look you can use it for now, as soon as I have a solution I'll let you know

le3tspeak, 2020-06-14:
is finished please download new version with the BUG Fix

closed 2020-06-14 by le3tspeak