le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #24: [BUG] (short description)

EG1313, 2020-06-19:
First of all - thank you for a working version of Marlin for Robin-Nano/Sapphire PRO.

Now... two issues.
First - all my gcodes end with below lines (end script) which perfectly works with "original" Sapphire firmware as well with Marlin 2.0 Bugfix but... for some reason it doesn't work with you version of 2.0.5.
It does everything except second and third lines - nozzle 2mm move up and moving entire gantry to a corner of X20 Y20 -WHY ?

G92 Z0
G1 Z2
G1 X20 Y20; 
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M106 S128; fan at 50%
G4 P45000; wait
M107; turn off cooling fan
M84; disable motors**

And the second "issue".
using Marlin 2.0 bugfix and your version of Marlin having vertical "ripples" when making circles/cylinders.
Tried with and without S-curve, different speed and jerks, linear advance on and off but... no success.
One could say that something is wrong with mechanics but... it never happens with original Two Trees firmware so no hardware is involved so... what is wrong here?

Many thanks again and appreciate any clues on all above.
P.S.  in all occasions has been used one/same gcode.

Attaching picture of "with and without ripple"

[image]
[image]

le3tspeak, 2020-06-19:
Which branch did you use it? If it wasn't the MKS-Robin-Nano I can't give you any support.

No help without expanding the configurations / pins files.

EG1313, 2020-06-19:
Hi, and thanks for coming back.

Yes, its MKS Robin nano version 1.1.
1. related to gcode - commands doesn't work when using the SD card but if I try to print from PC via USB - everything is fine.
2. related to a "ripple" - an hour ago, i have changed max speed, acceleration to a values identical to a "original" two trees firmware and the ripple diapered.

For now, I haven't changed anything in configuration.adv, only few things in configuration.h - attached.

Many thanks again and good luck

[Configuration.zip](https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/files/4806154/Configuration.zip)

le3tspeak, 2020-06-19:
But it is as I suspected you are using a very old version and not the current version with an update to the latest version, then all the errors where you wrote are fixed
Version 1.0.5b is the most current version of me

closed 2020-06-19 by le3tspeak

EG1313, 2020-06-20:
Hi!
first - thanks a lot, all my troubles are gone and printer BEHAVES :-)
Must say - impressed with elegant approach with various presets making it universal.
One thing... I've got this message on a very first boot, probably deserving attention (see attached picture).
Many thanks again and looking forward  for better versions.
Good luck!

[image]

le3tspeak, 2020-06-20:
Hi the message is normal as long as there is no EEPROM data on the printer, today version 1.0.6 Released I hope you have downloaded the latest :)