le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #25: [FR] Enable PID TUNING in Menu

thierryzoller, 2020-06-20:
Good experience with the Auto PID tuning, I recommend enabling it in Menu 👍 
#if ENABLED(PIDTEMP)
  #define PID_EDIT_MENU    
  #define PID_AUTOTUNE_MENU  
  //#define PID_PARAMS_PER_HOTEND // Uses separate PID parameters for each extruder (useful for mismatched extruders)

le3tspeak, 2020-06-20:
Is in the planning and coming has to be adjusted that you can set a temp is currently FIX set to 150 °

thierryzoller, 2020-06-20:
I think you may be mistaken, if you select the menu (set to 150) you can set it to any temperature you want. I did Pid tune to the temp I set :)

closed 2020-06-20 by le3tspeak

thierryzoller, 2020-06-21:
https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/commit/2113ea760a9f1ca38eb746d11f14f45f7ab4444e