le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #28: [FR] Tuning menu item with submenu items: PID Auto Tune and Calibrate E Steps

gtr33m, 2020-06-28:
### Description

I would like to have a "Tuning" menu item added to the main menu, with the submenu items:
* Perform PID Auto Tune
* Calibrate E-Steps

### Feature Workflow

1. Press Green button
2. Press Down to new menu item
3. Press Green button
4. Select function

Perform PID Auto Tune:
1. M303 E0 S200 C8
2. Screen displays "Performing PID Auto Tune"
3. Once complete screen displays "PID Auto Tune complete"

Perform E Steps Calibration:
1. Loop to preheat menu so user can select material
1. Screen displays "Measure and mark 120mm of filament", Wait for OK
2. send the command G1 E100 F100
3. Screen displays "Extruding 100mm of filament"
4. Screen displays "Measure distance to mark"
5. Allow user input from 0.0mm to 40mm
6. Read current E steps value from EEPROM
7. Calculate new E Steps value using formula: ([current E]*100)/(120-[user value])
8. Send command M92 E[new E steps value]
9. screen displays "E Steps tune complete"

#### Additional Information

Might be good to add a M500 command after each or prompt the user to save.

thierryzoller, 2020-06-28:
There is already a PID Autotune inbuilt ?

gtr33m, 2020-06-28:
> There is already a PID Autotune inbuilt ?

Yes, just no menu item to access it.

thierryzoller, 2020-06-28:
It is enabled and there is a menu in the latest build.

Here is the config file, search to "PID" : https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/blob/MKS-Robin-Nano/Marlin/Configuration.h

thierryzoller, 2020-06-28:
Partially duplicate of #25

gtr33m, 2020-06-28:
> It is enabled and there is a menu in the latest build.
> 
> Here is the config file, search for "PID" : https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/blob/MKS-Robin-Nano/Marlin/Configuration.h

Not in the menu as of 1.0.6 builds?

thierryzoller, 2020-06-28:
The menu is not intuitive to find - https://upam.uni.lu/content/download/3045/14922/file/LCD%20Menu%20Tree.pdf

gtr33m, 2020-06-28:
> The menu is not intuitive to find - https://upam.uni.lu/content/download/3045/14922/file/LCD%20Menu%20Tree.pdf

 #define PID_AUTOTUNE_MENU     // Add PID auto-tuning to the "Advanced Settings" menu. (~250 bytes of PROGMEM)

But it's not in my menu under Advanced Settings or any of the sub menus.

thierryzoller, 2020-06-28:
You don't have the latest version then:
https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/compare/1.0.6...MKS-Robin-Nano

gtr33m, 2020-06-28:
> You don't have the latest version then:
> [1.0.6...MKS-Robin-Nano](https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/compare/1.0.6...MKS-Robin-Nano)

[image]
[image]
[image]
[image]

But... this is in the middle of a print, so I'll have to double check when it's done.

thierryzoller, 2020-06-28:
PID Tuning is not accessible in the middle of a print ;)

gtr33m, 2020-06-28:
Checked after print and reboot. Menu is no different from above. Definitively not there. Checked configuration, advanced and temperature.

le3tspeak, 2020-06-28:
Is not activated in the release build 1.0.6 if you want it you have to build the firmware with the source code with VS code.

https://github.com/le3tspeak/Marlin-2.0.X-MKS-Robin-Nano/compare/1.0.6...MKS-Robin-Nano

closed 2020-06-28 by le3tspeak