le3tspeak/Marlin-2.0.X-MKS-Robin-Nano #30: PID_EXTRUSION_SCALING does not work ?

user32rtu, 2020-06-28:
Hello ! I have a question about "PID_EXTRUSION_SCALING".
When calibrating the bar feed, I found that the feed becomes smaller as the feed speed increases. I decided to compensate for the decrease in feed by increasing the temperature. I used "AUTOTEMP". With increasing print speed, I saw an increase in temperature. Then I turned off "AUTOTEMP" and turned on "PID_EXTRUSION_SCALING". With increasing print speed, the temperature did not change. I tried increasing "Kc = 150", but it did not give any effect.

le3tspeak, 2020-06-28:
PID_EXTRUSION_SCALING also does not raise the temperatures.
PID_EXTRUSION_SCALING is for adjusting the PID values during a print with nozzles over 0.6.

For temperatures is AUTO_TEMP

Linear Advance is used to adapt the lines.

closed 2020-06-28 by le3tspeak

user32rtu, 2020-06-28:
Thanks for the info.
Please tell me what is AUTOTEMP_OLDWEIGHT? How to use it?

thierryzoller, 2020-06-28:
You would need to ask Marlin for support, this is just a fork.

thierryzoller, 2020-06-28:
https://github.com/MarlinFirmware/Marlin/issues/11157